t-oster/LibLaserCut #30: Add lasercutting time estimation to other drivers

t-oster, 2015-09-22:
The Epilog driver support estimation of the job duration (https://github.com/t-oster/LibLaserCut/blob/develop/src/com/t_oster/liblasercut/drivers/EpilogCutter.java#L886).

I think the idea is the same for all lasercutters, so maybe we should port this method into the LaserCutter superclass, so every driver (i.e. LAOS and Smoothie) support this by default. Of cause the speed has to be configurable, so maybe this should come with a tutorial on how to calibrate this.

z4ziggy, 2016-03-27:
+1 for grbl :)

t-oster, 2016-11-21:
well... should not be that hard on g-code drivers, because they operate in fixed distance per time speeds rather than percent....

mgmax, 2020-01-25:
beta code: https://github.com/mgmax/LibLaserCut/commit/7a1dd7dc59eabaadf86405821ffcd15078bfa84a 

Will probably be merged together with #134 or some time after that. I don't want to tackle it before to avoid merge conflicts.

mgmax, 2023-12-24:
Current status: Time estimation is supported by most drivers. Can be added easily to new drivers, see https://github.com/mgmax/LibLaserCut/commit/0b63d63b119a399073217487eb8aa7f16533e372 .