t-oster/LibLaserCut #32: gcode export

mrzl, 2015-11-25:
hey,
amazing project(!) but let me get to the point:

is it possible to export gcode through the gui?

alternatively i suppose doing some modifications to GenericGcodeDriver.java by opening a BufferedWriter and adding additional output to it in sendLine()? then, i still can't trigger the process when no cutter is connected to my computer, because job execution is disabled until one is connected.

id be really happy about any hint! thanks already

t-oster, 2015-11-25:
Hi. Thanks for your interest. The GUI is in the VisiCut Project and your request is a duplicate of https://github.com/t-oster/VisiCut/issues/289. Any help appreciated. The task is easy, but I do not have time. I suggest a new File->Export menu in the UI and a saveJob method (as opposed to the sendJob method) in the LaserCutter superclass.

mrzl, 2015-11-30:
Thanks for the help @t-oster!

closed 2015-11-30 by mrzl