t-oster/LibLaserCut #45: java.lang.ClassCastException: java.lang.Float cannot be cast to java.lang.Integer

RincewindWizzard, 2016-02-26:
When using Visicut 1.7.9-23-g46d8904 there is an exception thrown, when calculating the duration of engraving. Seems like a problem with implicit/explicit casts. Should be fixed with:
`Integer intVal = (Float value).intValue();`
> java.lang.ClassCastException: java.lang.Float cannot be cast to java.lang.Integer
	at com.t_oster.liblasercut.PowerSpeedFocusProperty.setProperty(PowerSpeedFocusProperty.java:137)
	at com.t_oster.liblasercut.RasterPart.<init>(RasterPart.java:43)
	at com.t_oster.visicut.model.RasterProfile.addToLaserJob(RasterProfile.java:230)
	at com.t_oster.visicut.VisicutModel.prepareJob(VisicutModel.java:737)
	at com.t_oster.visicut.VisicutModel.estimateTime(VisicutModel.java:787)
	at com.t_oster.visicut.gui.MainView$66.run(MainView.java:2625)

t-oster, 2016-02-26:
Do you have any old lasercutter configfiles present? Please retry with cleared ".visicut" folder

madleech, 2016-02-28:
Oops that would be my fault. Fixed in PR #46

closed 2016-02-29 by t-oster

AbdelhadiBaben, 2017-12-29:
try : 
          int x = ((**Number**) floatVal).intValue();

mgmax, 2017-12-30:
Please provide some context: does the bug still occur for you with the newest VisiCut 1.8.x?

commit 1bc4cf4c (2016-02-29): Merge pull request #46 from madleech/fix-float-cast-error

Pass correct type to laser property (fixes #45)
commit 9c5bc7db (2016-02-29): Pass correct type to laser property (fixes #45)