t-oster/LibLaserCut #54: grbl & cheap laser engravers

z4ziggy, 2016-03-27:
The following is needed to work with latest grbl & the cheap chinese engravers:

the M03/M05 are needed to turn on/off the laser, otherwise it's kept turned on while moving.

the grbl fw doesn't reply with 'ok' but with a "Grbl 0.9i ['$' for help]" so I had to switch the throw exception with a simple print.

```
diff --git a/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java b/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java
index b52bff7..956edd4 100644
--- a/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java
+++ b/src/com/t_oster/liblasercut/drivers/GenericGcodeDriver.java
@@ -468,6 +468,7 @@ public class GenericGcodeDriver extends LaserCutter {
     currentSpeed = getTravel_speed();
     if (blankLaserDuringRapids)
     {
+      sendLine("M05", 0); // laser off
       currentPower = 0.0;
       sendLine("G0 X%f Y%f F%d S0", x, y, (int) (travel_speed));
     }
@@ -483,6 +484,7 @@ public class GenericGcodeDriver extends LaserCutter {
     String append = "";
     if (nextPower != currentPower)
     {
+      sendLine("M03", 0); //laser on
       append += String.format(FORMAT_LOCALE, " S%f", nextPower);
       currentPower = nextPower;
     }
@@ -532,7 +534,8 @@ public class GenericGcodeDriver extends LaserCutter {
       String line = waitForLine();
       if (!"ok".equals(line))
       {
-        throw new IOException("Lasercutter did not respond 'ok', but '"+line+"'instead.");
+        //throw new IOException("Lasercutter did not respond 'ok', but '"+line+"'instead.");
+        System.out.println("Lasercutter did not respond 'ok', but '"+line+"'instead.");
       }
     }
   }

```

pelrun, 2016-03-27:
I think you're doing something wrong.

Grbl should only be printing "Grbl 0.9i ['$' for help]" as it's initialisation message (which is detected elsewhere), NOT instead of 'ok'. And disabling the error is a kludge, not a fix.

Also, I believe raw Grbl really isn't the appropriate choice for a laser cutter; I'd seriously consider looking at the forks that are intended for this purpose.

z4ziggy, 2016-03-27:
tnx for the quick reply.
I agree regarding the "Grbl 0.9i ['$' for help]" - however, nothing i did was able to overcome the non-existing "ok" (the actual reply is a blank line) so removing the 'throw' was the *easiest* fix.

regarding replacing the grbl fw - i wasn't even aware there are other alternatives than Marlin. Any recommendations?

I Love VisiCut btw. Finally a laser engraving program that it is a compete drag & drop + WYSIWYG, without useless technical stuff.

t-oster, 2016-03-27:
Why didn't you just use the 'wait for ok' setting to disable the blocking?

t-oster, 2016-03-27:
... And why do you send Patches instead of pull requests?

pelrun, 2016-03-27:
Have a look at MrBeam (https://github.com/mrbeam/grbl), which is what @madleech is targetting with his grbl driver patches here.

z4ziggy, 2016-03-27:
seems like mrbeam requires a Mr Beam Shield. I'm trying to get the most out of the cheap engraver, hopefully a blog post to follow to assist others.

pelrun, 2016-03-27:
That's just the default pin configuration, it can be changed just like in standard grbl.

z4ziggy, 2016-03-27:
tnx for the advice. i'll try playing with this over the weekend.

madleech, 2016-03-28:
Hi @z4ziggy, I believe your issues are easily solved by:
* Using the latest development build of Visicut
* Using the GRBL driver instead of the generic gcode driver
* Enabling the "Force laser off during G0 moves" option

Using the GBRL specific driver will handle the reset etc much better.

If you want to use MrBeam, I'd suggest:
* Use the `mrbeam-stable` branch
* In `config.h` define your own `#define DEFAULTS_XYZ` define
* In `defaults.h` set up your own default values
* Adjust the pins in `cpu_map.h` under the `#ifdef CPU_MAP_ATMEGA328P` section

Out-of-the-box Grbl will work ok, however it can be quite jerky when engraving as each change in laser power causes the machine to stop moving momentarily. Using the MrBeam firmware nicely solves that issue giving silky-smooth engraving.

z4ziggy, 2016-03-28:
tnx @madleech .

I'm using latest dev Visicut branch, grbl driver (which is a subclass of gcode class), and blankLaserDuringRapids is 'true'. sending the M3/M5 was the only fix for me with default grbl fw.

as for the 2nd part of my ugly patch/hack, I agree with @t-oster that disabling waitForOKafterEachLine is a better fix, but I'm not as fluent in this code as you guys, so I wasn't aware of this being a config option.

I browsed mrbeam sources & looked at the config.h & default.h and already built it for my engraver, but I'll be free to upload & test the laser only over the weekend.

regarding the jerky movement using the default grbl fw - I was able to fix this with proper config to the STEPS_PER_MM & DEFAULT_ACCELERATION. Took me a while to get there, but it was worth it. I'm pretty amazed by the performance of the cheap engraver - the software was it's weakest link, and now VisiCut successfully closes that gap.

tnx again for your comment - I hope google results should display it for others looking into mrbeam & cheap laser engravers ;)

iceblu3710, 2016-11-22:
I just converted my K40 and the M3/M5 does not work if you keep your front panel current potentiometer as the L or K+ pin is the "laser Fire" and the wiper of the pot "IN" is the 0-5V current setting. If you keep the potentiometer in place the PWM is ignored when `BELOW` the pots set point so if the gcode send `G0 Xxx Yyy S0`  it will stay on to the pots set point during G0's

My wiring is:
`Spindle PWM => IN`
`Spindle DIR => L / K+`

This means:
`M3 = Laser On + PWM`
`M4 = Laser Off + PWM`
`M5 = Laser Off + PWM Off`

So my init gcode is `G21 G90 M3` and the closing gcode is `M5 G0 X0 Y0`

madleech, 2016-11-22:
There is a new Grbl release in the works which includes built in laser spindle support! https://github.com/gnea/grbl is where it is living for now. I'd suggest anyone trying to use a cheap Grbl powered laser engraver installs the 1.1 beta on their board for much better engraving.

iceblu3710, 2016-11-23:
Thanks for the tip, the lag between commands made things jerky sometimes. Will be the beta this weekend hopefully.

mgmax, 2017-05-14:
@z4ziggy It seems as if your problem with LibLaserCut is solved. Could you please close this bug report or state what exactly is the remaining problem and how it should be fixed? Thanks.

closed 2017-05-14 by z4ziggy