t-oster/LibLaserCut #65: Licensing incompatibility: GPL library present

pelrun, 2016-09-24:
In doing #64 I found the GLPK library used by SCPSolver is purely GPL licensed (which incidentally makes SCPSolver GPL licensed as well.)

This is incompatible with LLC's LGPL license.

Apparently if we switch from GLPK to LSOLVE (http://lpsolve.sourceforge.net/5.5/) then SCPSolver is also LGPL and the problem is resolved, although this requires testing.

pelrun, 2016-09-24:
This appears to only be used by the TSP optimiser.

mgmax, 2016-09-24:
Does anyone use the TSP optimizer? As far as I remember it was way too slow, and the "nearest" sorting strategy is good enough for most practical jobs.

I would therefore suggest removing it.

pelrun, 2016-09-25:
I don't disagree. https://github.com/t-oster/VisiCut/issues/226 essentially ended with the decision to pull it out, but it never happened.

If someone cares enough to make it work properly without the GLPK, then it can always be added back in.

pelrun, 2016-09-25:
I've added a commit removing the TSP and libraries to my PR now.

t-oster, 2016-09-26:
should be fixed then....

closed 2016-09-26 by t-oster