t-oster/LibLaserCut #67: G0 feed rate is always 3600

iceblu3710, 2016-11-19:
I have v1.7.9-62gfcede7b on ubuntu running with a grbl g-code output.

All G0 moves have F3600 even if the max feedrate is set <3600. It makes sense to have the G0 moves at max speed but this seems to be hard coded in the generic gcode driver.

````
 protected double travel_speed = 60*60;

  public double getTravel_speed()
  {
    return travel_speed;
  }
````

`setTravel_speed` is used to update that value but there is no call to do that it seems.

madleech, 2016-11-19:
Grbl ignores the feedrate on G0 rapid moves, so the travel speed setting has no effect. You can configure the travel speed on the Grbl board itself.

mgmax, 2016-11-19:
G0 commands should not have a F parameter, as it does not make sense for a "rapid move" command as opposed to the "controlled linear move" G1 command. If you want to obey a certain max. feedrate, just use G1.

See also: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g0

iceblu3710, 2016-11-19:
I agree that G0 commands should not bother outputting a feed rate then. Makes the gcode clear that the feed rate is up to something else. I will investigate if the new grbl v0.9j actually ignores feed rates on G0 though as my machine has 8000mm/s max feed rate and I can G1 around quicker it seems.

t-oster, 2016-11-21:
Ok, please tell me the result, so I can remove the speed option from G0 commands.

iceblu3710, 2016-11-22:
OK, the speed option is indeed ignored in G0 movements. 

I was just testing at G1 F1000 and Fmax=3000 which were similar enough I didn't even notice.

Stil shouldn't have speeds on G0 or at a minimum have it equal the machines stored max speed.

madleech, 2016-11-22:
Checked the Grbl source code and G0 commands ignore the feedrate: https://github.com/grbl/grbl/blob/658eb6a8b3cff0da59a61cbb5aa04d2c693c36c3/grbl/gcode.c#L966

Also tested on a real board and yep `G0 X100 F1` runs just as fast as `G0 X100 F10000`.

I've made a PR with this change.

iceblu3710, 2016-11-23:
Awesome, now I just need to get the source to compile on my machine and I will be in business.

mgmax, 2017-05-14:
Fixed by #71  -- please close this report. Thanks.

mgmax, 2018-12-26:
@t-oster please close this issue

closed 2018-12-26 by iceblu3710