t-oster/LibLaserCut #83: Joining of contiguous elements

Hueburtsonly, 2018-03-18:
There's a suboptimality with the way that Visicut/LibLaserCut handles INNER_FIRST optimization of certain input files where nominally closed loops are broken up into individual line segments, which are then cut according to Y position, which wastes a lot of time jogging about. I have implemented a fix, a pre-processing step in InnerFirstVectorOptimizer, which joins up elements with matching endpoints (and props) into larger elements -- most importantly, if a bunch of elements make a simple loop, it will make a single element which is the whole loop. This has two benefits:

* Jog time greatly reduced

* Watching the laser cutter, it actually _looks_ like an inner-first algorithm (i.e., it will avoid a lot of confusion and annoyance I've seen from people saying Visicut is "not doing what I asked" even though it sort of is, if only on a technicality)

Before starting a pull request, I wanted to ask first how you would like this to be presented to the user. Although the change will increase CPU time, the increase will be much smaller than the decrease I've introduced in #82. However, the memory usage will be considerably higher (don't have any hard numbers for you yet, but some constant factor more than current memory usage.) So, some options:

A. Simply modify INNER_FIRST to add the pre-processing.
B. Rename INNER_FIRST to INNER_FIRST_WITHOUT_LOOP_JOINING or INNER_FIRST_LITE or something like that, and add a new INNER_FIRST optimizer with the preprocessing step.
C. Add a new INNER_FIRST_WITH_LOOP_JOINING (or similar) optimizer.

I personally don't like C because I think 95% of users would be happy to see the new behaviour, and giving it a long intimidating name seems unfortunate. Take DELETE_DUPLICATE_PATHS as an example, I had no idea until I dug into the code that it uses NEAREST behind the scenes; I had always used NEAREST because I erroneously thought I had to make a choice between delete duplicates and having optimized pathing. So I'm in favour of option A, or B if we need to give users a way out in the so-far-hypothetical case of encountering memory issues with the pre-processing.

I don't claim that my change is the final say in how INNER_FIRST should be implemented, it's purely a step in the right direction (IMHO). The future work mentioned in comments inside InnerFirstVectorOptimizer is quite orthogonal to my change, but I'd also be happy to see my change replaced by a holistic solution.

> Is vectoroptimizers the right place to do this pre-processing when this is really a bug with the Visicut DXF importer/the DXF format*?

I think so.

* Even if there's a bug in the DXF importer used by Visicut, it's still possible that polylines can be split into lines in other software outside of Visicut's control, including imports coming in via other formats. Also, users might explicitly draw their parts using separate line segments.

* In any case, the joining cannot be done before the Mapping stage in Visicut, because it's possible that two segments that are otherwise compatible for joining end up not being joined because one gets Cut and the other is Marked, or skipped. So it would be inappropriate to add this fix as a naive post-import step. Given that the work therefore has to be done after "Execute" is clicked, it may as well be in with the optimizers.

mgmax, 2018-03-18:
I do not think the overhead is problematic because of the following experiment: Inkscape becomes slow if I duplicate a straight line about 9000 times, and almost unusable at about 50.000 times. With your improvements, exporting the G-Code with NEAREST sorting for 9000 lines is faster than the "select all" command in inkscape.

So if your code works up to 10k-50k lines in some seconds, I don't see any problems.
Therefore I'd say A. If B or C, then it should be in a way that existing settings using INNER_FIRST will use the new method.

See also https://github.com/t-oster/VisiCut/issues/203 for more thoughts on the algorithm and problematic corner cases. Most importantly, we do have to remove duplicates before joining, because otherwise the algorithm will create the same problems as now whenever a split path is duplicated.

For the 0,1% of cases where duplicate lines are explicitly needed, this should ideally be exposed to the UI via a checkbox (adding boolean flags for deduplication and joining to the VectorOptimizer class), but I think we can ignore this for now.

mgmax, 2018-03-18:
Cool, I'm looking forward to your changes. If it is not too complicated, it would be really great if you could implement some small tolerance (0.01mm or so) so that we aren't affected by floating point calculation errors.

pelrun, 2018-03-19:
Excellent! Doing this in inkscape is extremely inefficient (and requires a sequence of operations that's non-obvious) so having it done automatically is great.

Though I can think of several nasty edge cases that invariably cause problems when using a naive reconnection strategy that I've encountered multiple times when cleaning up an imported DXF - More than two path segments meeting at a point, and closely spaced parallel lines (i.e. closely spaced but critically distinct nodes) being the ones that immediately come to mind. Especially that last one, because if closely spaced nodes are joined together incorrectly at the vector optimisation stage there's no preview to warn you that the output is wrong.

I threw together a contrived example to demonstrate, unfortunately I don't have access to any real instances at the moment.

[hell.zip](https://github.com/t-oster/LibLaserCut/files/1825007/hell.zip)

rbowmaker, 2018-03-22:
Regarding hell.zip, the top two cases have always been fine, since #84 relies on perfect point equality, which isn't present in the top two cases (except the bounding box of the second case, which would get joined up and performed last, which is fine).

The third case would have caused issues with #84, but #86 will fix this.

mgmax, 2019-06-15:
As this issue is 99% fixed, and I haven't heard the usual user complaints anymore since #84, #86, I would suggest we close this, and open a new one for the remaining wishlist item: "Advanced vector optimization (deduplication, merge almost-equal points)".

Hueburtsonly, 2019-06-16:
I totally agree; although I'm happy with the current state of affairs, so I'll leave it to someone else to champion the cause and create a new bug for "Advanced vector optimization (deduplication, merge almost-equal points)" if they feel strongly about it. Thanks!

closed 2019-06-16 by Hueburtsonly