t-oster/LibLaserCut #87: certain paths fail to cut

FriedZombie, 2018-08-02:
In the case that a node overlaps at the starting point of the path in the svg, it creates a G0 and gcode with the same coordinates. Because the laserhead is already at the position previously set by the g0 command. The next G1 command is ignored. So the power of the laser stays zero.

It could be that the the issue is limited to only smoothy board, I don' t have a different board to test with.

May guess is that it would be something like checking if there are double travel points in the gcode.

[not cuttable.gcode](https://github.com/t-oster/LibLaserCut/files/2251539/not.cuttable.gcode.txt)
[not cuttable.svg](https://github.com/t-oster/LibLaserCut/files/2251540/not.cuttable.svg.txt)

t-oster, 2018-08-02:
Hi, what do you think the G-Code should look like? If I remove the G0 command, it would cut from (0/0) to (400/0), which would not be correct? So what do you suggest? Isn't this rather a bug in smoothieware?

FriedZombie, 2018-08-02:
True, but if you remove the first G1 command with the identical coordinates it would be fine, so instead of
'G0 X400.000000 Y0.000000 F10000 S0
G1 X400.000000 Y0.000000 S1.000000 F750
G1 X389.992400 Y0.000000'

it should be
'G0 X400.000000 Y0.000000 F10000 S0
G1 X389.992400 Y0.000000  S1.000000 F750'

Dropping the first G1 code with the coordinates identical to the G0 and append the other variables appended to the second one.

It can also be fixed within in smoothy board. although I think that more machines could have issues with it. Had the Tried it on a grbl avr chip, that doesn't like it either. (Then again smoothy is grbl based in its roots).

t-oster, 2018-08-02:
Please confirm that this is resolved in the latest version (build should finish in a few minutes)

closed 2018-08-02 by t-oster

FriedZombie, 2018-08-02:
Tested and works nicely, thank you very much

mgmax, 2018-08-02:
@FriedZombie you should nevertheless report this bug to the firmware developers, because it may reappear in odd cases (e.g. if the firmware rounds its input and two gcode coordinates are almost, but not exactly equal).

FriedZombie, 2018-08-03:
@mgmax I will report it it to grbl and smoothie as well, although I don't think it would be an issue. Even if of one coordinate the least significant digit is just incremented with 1 the G* command is not ignored by the firmware.

The reason that I wanted to have it fixed within visicut is because between different types of boards a lot of code is shared. So if it is also fixed in smoothie and grbl. There could be still a few out there that still would have that issue. (although I only have a smoothie board, I have looked trough some code of it though and found identical code in different boards).

commit 361273f1 (2018-08-02): Skip empty Segments. Fixes #87