t-oster/LibLaserCut #92: IllegalFormatConversionException in EpilogCutter.java

WieFel, 2018-09-19:
Hello,
I'm having an issue when trying to execute a simple job on an Epilog Helix cutter. My program looks as follows:
```
package it.fraunhofer.main;

import com.t_oster.liblasercut.FloatPowerSpeedFocusFrequencyProperty;
import com.t_oster.liblasercut.LaserJob;
import com.t_oster.liblasercut.PowerSpeedFocusFrequencyProperty;
import com.t_oster.liblasercut.VectorPart;
import com.t_oster.liblasercut.drivers.EpilogHelix;
import com.t_oster.liblasercut.platform.Util;

public class LaserCutterTest {

    public final static String IP_ADDRESS = "192.168.3.100";
    public final static int DPI = 300;

    public static void main(String[] args) throws Exception {
        System.out.println("Laser cutter test!");

        double width = Util.mm2px(10, DPI); // [mm]
        double height = Util.mm2px(10, DPI); // [mm]

        int power = 100;
        int speed = 15;
        int focus = 0;
        int frequency = 5000;

        // create EpilogHelix cutter instance
        EpilogHelix laserCutter = new EpilogHelix(IP_ADDRESS);

        PowerSpeedFocusFrequencyProperty property = new PowerSpeedFocusFrequencyProperty();
        property.setPower(power);
        property.setSpeed(speed);
        property.setFocus(focus);
        property.setFrequency(frequency);

        VectorPart vp = new VectorPart(property, DPI);
        vp.moveto(0, 0);
        vp.lineto(width, 0);
        vp.lineto(width, height);
        vp.lineto(0, height);
        vp.lineto(0, 0);

        LaserJob job = new LaserJob("CutRect", "Rect1", "Felix");
        job.addPart(vp);
        laserCutter.sendJob(job);

        System.out.println("Press START on the laser cutter!");
    }

}
```

When sending the job to the laser cutter, an exception within the `generateDummyRaster()` method occurs. Here is the stacktrace:

```
Exception in thread "main" java.util.IllegalFormatConversionException: d != java.lang.Double
	at java.base/java.util.Formatter$FormatSpecifier.failConversion(Formatter.java:4421)
	at java.base/java.util.Formatter$FormatSpecifier.printInteger(Formatter.java:2936)
	at java.base/java.util.Formatter$FormatSpecifier.print(Formatter.java:2890)
	at java.base/java.util.Formatter.format(Formatter.java:2671)
	at java.base/java.io.PrintStream.format(PrintStream.java:1053)
	at java.base/java.io.PrintStream.printf(PrintStream.java:949)
	at com.t_oster.liblasercut.drivers.EpilogCutter.generateDummyRaster(EpilogCutter.java:563)
	at com.t_oster.liblasercut.drivers.EpilogCutter.generatePjlData(EpilogCutter.java:773)
	at com.t_oster.liblasercut.drivers.EpilogCutter.realSendJob(EpilogCutter.java:334)
	at com.t_oster.liblasercut.drivers.EpilogHelix.realSendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendJob(EpilogCutter.java:400)
	at com.t_oster.liblasercut.drivers.EpilogHelix.sendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.LaserCutter.sendJob(LaserCutter.java:107)
	at com.t_oster.liblasercut.LaserCutter.sendJob(LaserCutter.java:124)
	at it.fraunhofer.main.LaserCutterTest.main(LaserCutterTest.java:51)
```

Any solutions for this?
Thank you

t-oster, 2018-09-19:
Looks Like here EpilogCutter.java:563 is a "d" which should be an "i"?

WieFel, 2018-09-20:
Thank you.
I changed the lines 564 and 565 of EpilogCutter.java from
```
out.printf("\033*r%dT", jp.getMaxY());//height);
out.printf("\033*r%dS", jp.getMaxX());//width);
```
to
```
out.printf("\033*r%iT", jp.getMaxY());//height);
out.printf("\033*r%iS", jp.getMaxX());//width);
```
Now I get a different error:
```
Exception in thread "main" java.util.UnknownFormatConversionException: Conversion = 'i'
	at java.base/java.util.Formatter$FormatSpecifier.conversion(Formatter.java:2837)
	at java.base/java.util.Formatter$FormatSpecifier.<init>(Formatter.java:2863)
	at java.base/java.util.Formatter.parse(Formatter.java:2711)
	at java.base/java.util.Formatter.format(Formatter.java:2653)
	at java.base/java.io.PrintStream.format(PrintStream.java:1053)
	at java.base/java.io.PrintStream.printf(PrintStream.java:949)
	at com.t_oster.liblasercut.drivers.EpilogCutter.generateDummyRaster(EpilogCutter.java:564)
	at com.t_oster.liblasercut.drivers.EpilogCutter.generatePjlData(EpilogCutter.java:774)
	at com.t_oster.liblasercut.drivers.EpilogCutter.realSendJob(EpilogCutter.java:334)
	at com.t_oster.liblasercut.drivers.EpilogHelix.realSendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.drivers.EpilogCutter.sendJob(EpilogCutter.java:400)
	at com.t_oster.liblasercut.drivers.EpilogHelix.sendJob(EpilogHelix.java:35)
	at com.t_oster.liblasercut.LaserCutter.sendJob(LaserCutter.java:107)
	at com.t_oster.liblasercut.LaserCutter.sendJob(LaserCutter.java:124)
	at it.fraunhofer.main.LaserCutterTest.main(LaserCutterTest.java:46)
```

But still, I think there might be some other problem. The return types of `jp.getMaxY()` and `jp.getMaxX()` are double, which should be ok with the `%d`.

Any other suggestions?

t-oster, 2018-09-20:
No it should be '%f' or cast the values to int

t-oster, 2018-09-20:
@mgmax looks like the commit https://github.com/t-oster/LibLaserCut/commit/087c2667002d8ffcd49833c4ebbef2f4b2520dfe has changed the maxX and Y to double, which may not make sense in a raster part?

t-oster, 2018-09-20:
@WieFel please change the files to 
```java
    out.printf("\033*r%dT", (int) jp.getMaxY());//height);
    out.printf("\033*r%dS", (int) jp.getMaxX());//width);
```
for now. I have to investigate all consequences before making any assumptions.

t-oster, 2018-09-20:
there are a few other lines. See this patch:
```patch
--- a/src/com/t_oster/liblasercut/drivers/EpilogCutter.java
+++ b/src/com/t_oster/liblasercut/drivers/EpilogCutter.java
@@ -560,8 +560,8 @@
     /* Focus */
     out.printf("\033&y%dA", mm2focus(prop.getFocus()));
 
-    out.printf("\033*r%dT", jp.getMaxY());//height);
-    out.printf("\033*r%dS", jp.getMaxX());//width);
+    out.printf("\033*r%dT", (int) jp.getMaxY());//height);
+    out.printf("\033*r%dS", (int) jp.getMaxX());//width);
         /* Raster compression:
      *  2 = TIFF encoding
      *  7 = TIFF encoding, 3d-mode,
@@ -737,18 +737,18 @@
           }
           case MOVETO:
           {
-            out.printf("PU%d,%d;", cmd.getX(), cmd.getY());
+            out.printf("PU%d,%d;", (int) cmd.getX(), (int) cmd.getY());
             break;
           }
           case LINETO:
           {
             if (lastType == null || lastType != VectorCommand.CmdType.LINETO)
             {
-              out.printf("PD%d,%d", cmd.getX(), cmd.getY());
+              out.printf("PD%d,%d", (int) cmd.getX(), (int) cmd.getY());
             }
             else
             {
-              out.printf(",%d,%d", cmd.getX(), cmd.getY());
+              out.printf(",%d,%d", (int) cmd.getX(), (int) cmd.getY());
             }
             break;
           }
```

WieFel, 2018-09-21:
Ok that solved the problem, thank you!

PR #93 (2018-09-23): Fix: Epilog-Cutter uses int based coordinates (more fixes)
This fixes #92 also for engraving

The problem was a regression caused by the change to double coordinates in 087c2667002d8ffcd49833c4ebbef2f4b2520dfe
closed 2018-09-21 by WieFel