t-oster/LibLaserCut #121: Add PPI power control to stock K40 Drivers.

tatarize, 2019-11-06:
After writing the K40 driver for this project I started up meerK40t (https://github.com/meerk40t/meerk40t) which is it's own K40 laser software in python. During development though I realized there actually is a way to add PPI power control to the stock K40. 

https://github.com/meerk40t/meerk40t/wiki/How-does-MeerK40t's-pulse-modulation-works%3F

This driver upgrade would also give me a chance to test the settings changes @t-oster said existed in t-oster/VisiCut#515. When it was merely a question changing an already functional thing it wasn't highly worth it, but given that power modulation for the stock K40 is a pretty killer feature, it actually should be totally worth it.

tatarize, 2020-01-29:
@t-oster I checked and the way the K40Driver work in Windows it requires LibUsb to have a driver entry in in the control panel.  This tends to be how Whisperer works (by far the most popular program for the stock controller (circa Jan. 2020) and MeerK40t (up to 0.4.0). I checked through the driver source-code etc, and I've managed to make methods to use the native windows driver. Other OSes will typically work fine with the Libusb, and they will not work with the Windows only specific driver. 

Implementing the specific windows driver would require JNA to access the DLL file. This is already a dependency of Visicut but is not a dependency of LibLaserCut. In theory I should be able to just throw/catch it as a dependency fallback.

Should I add this to the todo list? It's likely not too far from the beaten path if I update JK40. And I've more than finished that stuff for my work in MeerK40t.

t-oster, 2020-01-29:
I guess we could add JNA as a dependency for LibLaserCut. Maybe we could even integrate the dll into the jar so it works without manually installing, but that needs careful testing.

tatarize, 2020-01-29:
It automatically installs with windows. And might need to be installed as it's properly a driver. The libusb thing doesn't work without specifically being installed so I daresay the rest are the same way. But, for windows when you plug-in the laser it's the driver that automatically installs.

t-oster, 2020-01-29:
Ok as long as we make sure visicut still works without that driver installed (for other lasercutters) it would be a great addition.

tatarize, 2020-01-29:
Guaranteeing that requires a number of tests:
1. Windows (No CH341 driver, No LibUsb driver) -- Should fail.
2. Windows (Default CH341 driver, No LibUsb driver) -- Should work.
3. Windows (No CH341 driver, LibUsb driver) -- Should work.
4. Linux (No explicit drivers) - Should use bundled Libusb and work.
5. OSX (No explicit drivers) - Should use bundled Libusb and work.

The change, in theory, would make case 2 work when it currently should fail. I don't know that I checked it super-explicitly.  But, as Non-HID devices in windows it might well need a libusb installed driver akin to how Whisperer uses Zadig. I know it was a major hassle for Scorch which is why K40 Whisperer is packaged with the libusb driver. I've removed that limitation for some of the software I work on. And should likely do the same here.
https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows

That and I've been meaning to backport the power modulation I developed into this project.

tatarize, 2020-01-30:
The JNA is being rude and with no error messages it's really hard to debug. Need to do the dll translations and it just outright rejects for an unknown reason. I'm not an expert in JNA. Very similar to working code exists:

https://github.com/K40Nano/Jk40/tree/tatarize-ppi/jk40
 
But, I'm a bit stumped. Getting the PPI though, that's just an algorithm change. But, testing requires visicut to compile and I didn't seem to get that to work.

tatarize, 2020-02-23:
This issue is resolved with the merge of #136 

The only holdup is my windows machine decides to hard crash, on the hook grab from the USB.

In theory this code should be the same as it was before. However, it still lacks a real world test. I guess I could do so on my RPi. Though windows would likely be the more standard platform people use.

closed 2020-04-02 by t-oster