t-oster/LibLaserCut #123: Consider making https://github.com/LaosLaser less prominent on the list

matkoniecz, 2019-11-07:
LAOS seems to be a dead project, with last traces of a commit activity at https://github.com/LaosLaser in 2017

t-oster, 2019-11-07:
Not sure. I know there are several people using a LAOS laser. What list are you talking about?

matkoniecz, 2019-11-07:
in https://github.com/t-oster/LibLaserCut README

In my case I started going through the list trying to find match, and first one was seemingly dead LAOS project.

I am not proposing removing it, just moving to the end of the list so first encounter will not be "oh, this is old project supporting dead projects".

BTW, would be OK to move README to Markdown? It would be basically change of extension, removal of newlines (if acceptable), and should cause it to render better on Github.

t-oster, 2019-11-07:
of cause. Markdown is used everywhere else. The list has no special order, feel free to reorder and improve it.

commit a4e79fd8 (2019-12-27): Re-Sort list of supported devices

Fixes #123
closed 2019-12-27 by mgmax