t-oster/LibLaserCut #126: Debugging with maven

mgmax, 2019-11-14:
How can I debug LibLaserCut (breakpoint, single-step, etc.) within VisiCut in Netbeans? Since switching to maven it doesn't seem to work anymore. Or am I doing something wrong?

t-oster, 2019-11-14:
If you have both projects open netbeans should recognize that and allow debugging.... I will check tomorrow

mgmax, 2019-11-14:
After cleaning and closing everything, now it works again. Possible reasons include:
- problem is between keyboard and chair
- it depends on the order in which you open projects
- NetBeans is confused if you open the directory via a symlink
Now it works.

closed 2019-11-14 by mgmax

t-oster, 2019-11-14:
Yes, I also checked it and it worked. I opened VisiCut and then LibLaserCut which is checked out in the same module.However the connection should work no matter where the projects are located, since NetBeans just detects that VisiCut depends on LibLaserCut with a specific version which is also an open project. Maybe you had an old "nbproject" folder in the project? With maven we do not need any netbeans specific project configuration anymore, so your folder should not need any extra files.