t-oster/LibLaserCut #129: building VisiCut using make - fails saying libLaserCut version 11 is wrong

dlarue, 2019-12-24:
on Ubuntu 18.04.3 with everything but netbeans installed and it fails "make jar" showing what looks a failed request for building version 11 of LibLaserCut. Looking in the pom.xml file in LibLaserCut it does state version 11 on line 172.

build output:
(base) dlarue@Lenovo-Z70-80:~/Projects/VisiCut$ make jar
./generatesplash.sh
Checking for rsvg...
found rsvg-convert.
Determining Version (may be overridden with environment variable VERSION)
Version is: ""
Version is: 
Generating SVG
Converting to png
cleaning...
done.
test -f LibLaserCut/pom.xml  || { echo "Error: the LibLaserCut submodule is missing. Try running 'git submodule update --init'."; false; }
cd LibLaserCut && mvn install
[INFO] Scanning for projects...
[INFO] 
[INFO] --------------------< de.thomas-oster:liblasercut >---------------------
[INFO] Building liblasercut 1.9-SNAPSHOT
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-resources-plugin:2.6:resources (default-resources) @ liblasercut ---
[INFO] Using 'UTF-8' encoding to copy filtered resources.
[INFO] Copying 1 resource
[INFO] 
[INFO] --- maven-compiler-plugin:3.1:compile (default-compile) @ liblasercut ---
[INFO] Changes detected - recompiling the module!
[INFO] Compiling 78 source files to /home/dlarue/Projects/VisiCut/LibLaserCut/target/classes
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  1.055 s
[INFO] Finished at: 2019-12-24T09:13:52-08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.1:compile (default-compile) on project liblasercut: Fatal error compiling: invalid target release: 11 -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
Makefile:24: recipe for target 'libLaserCut' failed
make: *** [libLaserCut] Error 1

mgmax, 2019-12-24:
Please try installing openjdk-11-jdk.

dlarue, 2019-12-24:
I think I figured it out using mvn --version

It wanted Java JDK version 1.11 instead of the Java JRE version 1.8 I had defaulted JAVA_HOME to.
Correcting my JAVA_HOME to point to java-1.11.0-openjdk-amd64 worked.

Wow, you are fast @mgmax   Thanks.  I just saw your post as I was about to hit the close/comment button.

closed 2019-12-24 by dlarue