t-oster/LibLaserCut #131: Source port control in commons-net-tftp breaks LAOS

dirkx, 2019-12-29:
We needed to make this https://github.com/MakerSpaceLeiden/commons-net/commit/5c3b9228135e07fe366afbdf5bde73982ad16eaf change in commons.net to keep visicut working with our LAOS laser cutters (https://wiki.makerspaceleiden.nl/mediawiki/index.php/Laser_Engraver).

Dw.

t-oster, 2019-12-29:
It seems as if this was implemented because of this bug report http://apache-commons.680414.n4.nabble.com/jira-Created-NET-414-Apache-Commons-TFTP-does-not-reject-request-replies-that-originate-from-a-contr-td3566898.html

dirkx, 2019-12-31:
Hmm - the issue seems to be that pre 1990 code and the world+dog in IoT uses the standard https://github.com/ARMmbed/lwip/blob/master/src/apps/tftp/tftp_server.c example or some variation thereof.

mgmax, 2020-01-25:
@dirkx That's also my interpretation of the RFC -- can you submit a pull request for commons-net?

mgmax, 2020-02-19:
@dirkx Please try to submit a pull request for commons-net so that we don't need to fork it.

t-oster, 2020-03-25:
Seems to be fixed in LAOS https://github.com/LaosLaser/Firmware/issues/63#issuecomment-603732692. @dirkx Can we close this?

mgmax, 2020-09-16:
The issue is only partially solved. There is a LAOS firmware update, but of course VisiCut should also work with the older firmware.
Unfortunately the original poster @dirkx  fails to respond to any of my questions, though we almost have the solution at hand. Maybe someone else wants to try?