t-oster/LibLaserCut #137: Pass color information from SVG to driver

kkaempf, 2020-01-30:
## Motivation

Some Lasercutters, like those using the [Ruida controllers](https://www.cloudraylaser.com/collections/ruida-controller), have the ability to preview the laser path in a small color display.

Since VisiCut maps SVG colors to laser settings, it would be helpful to visualize the same colors in this display. Drivers like [ThunderLaser](https://github.com/kkaempf/LibLaserCut/tree/ThunderLaser_Ruida) currently use random colors to visually differentiate layers and their power settings.

## Goal

Pass the `rgb` color information from the input `.svg` file to the `LaserProperty` object attached to the set of `VectorCommands` representing the respective svg path elements.

Drivers can then use this to enrich the laser cutter commands and display a color-correct preview in displays.

t-oster, 2020-01-30:
Well I am not sure if this is the right approach. Visicut can map any svg property (color,layer,thickness etc) to a Laser-Profile like cut,mark cut deep or whatever you like. And the preview shows the color of the laser profile. Eg I use to draw cut in red end mark (=cut but not cut through) in black. But I can have a mapping which maps blue and green paths to cut and everything else to mark, so I would rather use the color of the Laser-Profile. The it would match the preview in visicut and it would also work if I have a layer for cut and another layer for mark.

kkaempf, 2020-01-30:
>  it would match the preview in visicut 

:+1:

mgmax, 2020-01-30:
However, the profiles in VisiCut don't have a preview color. There are three colors per *material*: engrave, cut and background. Line profiles can be switched between the cutting color ("cut through") and the engrave color ("engrave line").

t-oster, 2020-01-30:
@mgmax you are right... I should use my own program more often....