t-oster/LibLaserCut #140: Integration testing against known-good output

mgmax, 2020-02-05:
Currently, it is easy to break some driver without anyone noticing.
We should have a set of input jobs and known (machine-tested) output files for each driver.
The CI testing should then check that LibLaserCut still produces the same output file. On intentional changes, the set of known-good outputs must be regenerated, and then we know that a specific change should be tested on real machines.

t-oster, 2020-02-06:
Good idea. I think we should create a unit test which iterates a specific folder which contains those files, maybe in subfolders named like the driver classes

t-oster, 2020-02-06:
However you would have to do this in visicut since liblasercut doesn't know any input file format

mgmax, 2020-02-06:
Hm. On the other hand, testing with VisiCut could be quite fragile, as we depend on exactly the same rendering, and we will only notice once the LibLaserCut submodule has been updated.

Testing in LibLaserCut should be possible with LaserScript or just a plain java file such as https://github.com/t-oster/LibLaserCut/blob/d6728e8df802d631fd9f1e54d73fa45a2a26d04b/src/test/java/de/thomas_oster/liblasercut/drivers/LaosCutterTest.java . The only technical requirement would be that the driver supports "write to file". The downside would be that sending the exact test file to the real machine wouldn't be so easy because you'd have to set the IP etc. in the source code.

However, since most (all?) of these bugs were introduced by changes that should not have changed anything in the machine output, I think that a quick-and-dirty test in LibLaserCut would still be a good compromise between development effort and benefit. In the rare case that something changed, then we at least know that testing with VisiCut (or careful analysis of the diff) is required.

t-oster, 2020-02-07:
We could modify visicut to have an option to write the Laserjob object eg as json to some file, which would be the equivalent of a driver's input file. Then we can create the test jobs easily with visicut and just load the Laserjob object in the unit test

t-oster, 2020-02-07:
The plain Java way also works but engraving seems to be hard to write, we would need to import some raster image

phdoerfler, 2020-03-08:
I'll take a crack at this.

mgmax, 2020-03-08:
@phdoerfler my first attempt I did inbetween can be found in #145. Can certainly be improved.

commit fe743b5a (2020-03-10): Automated test that driver output doesn't change

Only supports drivers that implement saveJob().

Closes #140
closed 2020-03-10 by mgmax