t-oster/LibLaserCut #146: Enable Codacy for LibLaserCut

phdoerfler, 2020-03-09:
[Codacy](http://codacy.com) is a free (for open source projects like this) webapp that performs static code analysis and tells you how well a project is doing in terms of code quality.

I have it enabled on my forks of LibLaserCut and Visicut and it highlights quite a number of issues worth tackling.

mgmax, 2020-03-13:
seems done

mgmax, 2020-04-04:
@t-oster Codacy doesn't seem to be active anymore, at least it doesn't show up in the pull requests, for example #148.

t-oster, 2020-04-04:
Looks as if the repository has to belong to an organization in order for Codacy to work correctly

mgmax, 2020-04-04:
But strangely enough it worked some time ago: 
Here it works:
[image]
Here it doesn't show up:
[image]

phdoerfler, 2020-04-09:
Can we also add the Codacy badge to the README? I would have filed a PR but I'm afraid that's something a repo admin has to do.

t-oster, 2020-04-20:
I added the badges. Not sure if everything works now, but in codacy I can see the commits.

closed 2020-05-13 by t-oster