t-oster/LibLaserCut #159: Windows Compile Fails AllDriversTest, by Unix Line-end.

tatarize, 2020-05-13:
Windows has issues compiling LibLaserCut because it fails the All Drivers Test. This is because the line endings in Windows are different than the line ending in Linux. The `K40StockDriver` and `IModelaMill` fail, and some others that use `\n` in their code production.

mgmax, 2020-05-13:
Are the generated files different, or the git checkout? On Linux, `de.thomas_oster.liblasercut.drivers.IModelaMill.out` and the `.out.new` file both have `\n` only.
If I understand the documentation of [gitattributes](https://www.git-scm.com/docs/gitattributes) correctly, then the `.gitattributes` file in /test-output should prevent any newline changes by `git`.

tatarize, 2020-05-14:
The generated files are different. They don't match the downloaded version. It sends out `\n` and windows calls that `\r\n` and not just `\n`.

tatarize, 2020-05-14:
> Tests run: 1, Failures: 0, Errors: 1, Skipped: 0, Time elapsed: 1.38 sec <<< FAILURE!
> compareWithKnownOutput(de.thomas_oster.liblasercut.drivers.AllDriversTest)  Time elapsed: 1.179 sec  <<< ERROR!
> java.lang.Exception: Driver output changed:
> Output for de.thomas_oster.liblasercut.drivers.IModelaMill changed. For details, compare the files .\test-output\de.thomas_oster.liblasercut.drivers.IModelaMill.out and .\test-output\de.thomas_oster.liblasercut.drivers.IModelaMill.out.new manually. If this change is okay, delete the old file .\test-output\de.thomas_oster.liblasercut.drivers.IModelaMill.out , rerun the tests twice and then don't forget to commit the changed file.
> Output for de.thomas_oster.liblasercut.drivers.K40NanoDriver changed. For details, compare the files .\test-output\de.thomas_oster.liblasercut.drivers.K40NanoDriver.out and .\test-output\de.thomas_oster.liblasercut.drivers.K40NanoDriver.out.new manually. If this change is okay, delete the old file .\test-output\de.thomas_oster.liblasercut.drivers.K40NanoDriver.out , rerun the tests twice and then don't forget to commit the changed file.
> 
> 	at de.thomas_oster.liblasercut.drivers.AllDriversTest.compareWithKnownOutput(AllDriversTest.java:163)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
...

From a clean version just taken  into windows and told to test.

tatarize, 2020-05-14:
[test-output.zip](https://github.com/t-oster/LibLaserCut/files/4626117/test-output.zip)

mgmax, 2020-05-27:
Thanks, it was really a bug that a different line ending was sent to the cutter (or stored when saving files) on Linux and Windows for these two cutters.

commit 275ca600 (2020-05-27): Fix #159: Avoid PrintStream, use LF as line separator on all platforms

Now the tests should run correctly on Windows
closed 2020-05-27 by mgmax