t-oster/LibLaserCut #164: How to archive checkboxes in the "Edit Laser" Dialog

ahorn42, 2020-08-12:
What is necessary to get a property that shows a chekbox like highlighted in the image below?

I had a look at `GenericGcodeDriver.java` and did the same to my property (at least in my opinion ;) ), but there is no checkbox shown at my entry. 

[image]

My setters an getters for my properties:
```java
protected static final String SETTING_PEN_DROP_DISTANCE = "Pen drop distance (in mm)";
protected static final String SETTING_LIFT_PEN_AFTER_EVERY_LINE = "Lift pen after every line";

protected Double penDropDistance = 10.0;

public Double getPenDropDistance() {
  return penDropDistance;
}

public void setPenDropDistance(Double penDropDistance) {
  this.penDropDistance = penDropDistance;
}

protected Boolean liftPenAfterEveryLine = false;

public Boolean getLiftPenAfterEveryLine() {
  return liftPenAfterEveryLine;
}

public void setLiftPenAfterEveryLine(Boolean liftPenAfterEveryLine) {
  this.liftPenAfterEveryLine = liftPenAfterEveryLine;
}

@Override
public Object getProperty(String attribute) {
  Object result = super.getProperty(attribute);
  if (result == null) {
    if (SETTING_PEN_DROP_DISTANCE.equals(attribute)) {
      this.getPenDropDistance();
    } else if (SETTING_LIFT_PEN_AFTER_EVERY_LINE.equals(attribute)) {
      this.getLiftPenAfterEveryLine();
    }
  }
  return result;
}

@Override
public void setProperty(String attribute, Object value) {
  super.setProperty(attribute, value);
  if (SETTING_PEN_DROP_DISTANCE.equals(attribute)) {
    this.setPenDropDistance((Double) value);
  } else if (SETTING_LIFT_PEN_AFTER_EVERY_LINE.equals(attribute)) {
    this.setLiftPenAfterEveryLine((Boolean) value);
  }
}
```

`SETTING_LIFT_PEN_AFTER_EVERY_LINE` is the property which should show a checkbox.

mgmax, 2020-08-12:
The property name must be included in `getPropertyKeys()`, for example:
https://github.com/t-oster/LibLaserCut/blob/5286f27c07f8ee03581d500ffd954f22c668ff31/src/main/java/de/thomas_oster/liblasercut/drivers/Grbl.java#L60-L73

ahorn42, 2020-08-12:
Oh - I forgot to post this method as well. I did this. The property is indeed visible in the panel, but there is no Checkbox, but a regular text input, although it is a Boolean.

```
@Override
  public String[] getPropertyKeys() {
    List<String> result = new LinkedList<>(Arrays.asList(super.getPropertyKeys()));
    result.remove(GenericGcodeDriver.SETTING_FLIP_X);
    result.remove(GenericGcodeDriver.SETTING_FLIP_Y);
    result.add(DexArm.SETTING_PEN_DROP_DISTANCE);
    result.add(DexArm.SETTING_LIFT_PEN_AFTER_EVERY_LINE);
    return result.toArray(new String[0]);
  }
```

mgmax, 2020-08-12:
`getProperty()` is not correctly returning the value, `this.getPenDropDistance()` should probably be `return this.getPenDropDistance()`. (Also, then you might have to delete the preferences folder.)

ahorn42, 2020-08-12:
argh ... Thanks! And sorry for that issue ;D 

sometimes one is blind :see_no_evil:

closed 2020-08-12 by ahorn42