t-oster/LibLaserCut #170: Win 10: 'Underlying input stream returned zero bytes'

darchivist, 2021-04-05:
Trying to send to a Wainlux K6 on COM3. I’ve not been able to send a simple pattern the the engraver; it errors out with the above message. I was able to get it to trace the bounding box once by running the app in admin mode; I’ve not been able to get this to repeat. 

The laser’s native app (such as it is) works fine on the same COM port. I’ve rebooted the system between attempts.

mgmax, 2021-04-05:
Which laser driver do you use?

Please try running VisiCut [from the command line](https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline) and post a screenshot of the detailed error message.

mgmax, 2021-04-05:
We had a similar issue previously, where the error message indicated a timeout. https://github.com/t-oster/LibLaserCut/pull/25 

Maybe the driver does not correctly set the serial mode (baud rate, flow control, parity and so on), and only randomly works if you used the native app before so that the serial mode is already set?

darchivist, 2021-04-05:
I’m using the ‘K3/K5 Laser Engraver’ on the dropdown; baud is set to 115200.

I’ll try running from the command line and post the message.  Thanks for the help!

[EXTERNAL]

darchivist, 2021-04-05:
Interesting thought; with the poorly-documented Wainlux K6, I wouldn’t be surprised.  I’ll see what I can puzzle out.

Thanks!

[EXTERNAL]

darchivist, 2021-04-06:
Debug results:

WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.dom4j.io.SAXContentHandler (file:/C:/Program%20Files%20(x86)/VisiCut/Visicut.jar) to method com.sun.org.apache.xerces.internal.parsers.AbstractSAXParser$LocatorProxy.getEncoding()
WARNING: Please consider reporting this to the maintainers of org.dom4j.io.SAXContentHandler
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
using single-instance port: 6546
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
java.io.IOException: Underlying input stream returned zero bytes
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.Reader.read(Unknown Source)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.waitForACK(K3EngraverDriver.java:336)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendLine(K3EngraverDriver.java:353)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.moveRelative(K3EngraverDriver.java:507)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendJob(K3EngraverDriver.java:669)
        at de.thomas_oster.visicut.VisicutModel.sendJob(VisicutModel.java:778)
        at de.thomas_oster.visicut.gui.MainView.lambda$executeOrSaveJob$7(MainView.java:2209)
        at java.base/java.lang.Thread.run(Unknown Source)
java.io.IOException: Underlying input stream returned zero bytes
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.Reader.read(Unknown Source)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.waitForACK(K3EngraverDriver.java:336)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendLine(K3EngraverDriver.java:353)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.moveRelative(K3EngraverDriver.java:507)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendJob(K3EngraverDriver.java:669)
        at de.thomas_oster.visicut.VisicutModel.sendJob(VisicutModel.java:778)
        at de.thomas_oster.visicut.gui.MainView.lambda$executeOrSaveJob$7(MainView.java:2209)

Madz83, 2021-07-23:
I have the same issue. I've installed the Driver that came with  Wainlux K6 and then installed the latest Visicut software. It gives the same error as highlighted above. Please help. 
I'm attaching screenshots of port configuration and error message from Visicut. 
[image]
[image]

mgmax, 2021-07-24:
VisiCut seems to get stuck waiting for a response from the laser cutter. There are two possible reasons:

1. We are using the wrong baud rate / handshake setting
2. The protocol is different

1.

Does it work if you first laser something with the manufacturer's app and then with VisiCut?

What are the serial port settings *while/after* using the manufacturer's app?
(Device manager - right click - properties - port settings)

2.

Can you record a trace of how the manufacturer's app communicates with the laser, compared to how VisiCut does it? I'm not sure how exactly this works on Windows, you may try the following links. (Disclaimer: I have not tested any of these)

https://lorgor.blogspot.com/2010/01/tracing-serial-port-on-windows.html
http://www.simpleterm.co.uk/site/
https://www.eltima.com/products/serial-port-monitor/

t-oster, 2021-07-27:
@Madz83 can you provide a log from running VisiCut via command line (https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline) and try to just vector-cut a simple shape? I would like to see where exactly the timeout occurs.

DokH, 2021-11-28:
Attached COM port traces (using https://www.eltima.com/products/serial-port-monitor/) when using application from vendor and using Visicut.
Hope that helps.
[Wainlux K6 - COM trace.zip](https://github.com/t-oster/LibLaserCut/files/7613640/Wainlux.K6.-.COM.trace.zip)

DokH, 2021-11-28:
> @Madz83 can you provide a log from running VisiCut via command line (https://github.com/t-oster/VisiCut/wiki/Development:-Running-from-commandline) and try to just vector-cut a simple shape? I would like to see where exactly the timeout occurs.

See debug posted by @darchivist (https://github.com/t-oster/LibLaserCut/issues/170#issuecomment-813736272).
That is all the debug output we get.

mgmax, 2022-08-12:
I can't open that "COM trace" file on Linux, maybe you can post a screenshot of the start of the communication?

Please also consider the questions about "1. We are using the wrong baud rate / handshake setting" I asked earlier. Maybe the COM trace will also show us which settings were used. Or at worst we would have to use a logic analyzer on the to look at the communication on the hardware level.

The question is still to find out what VisiCut does different than the original software, to find out at which point the communication fails.

JarrettR, 2022-08-12:
Also getting this. Windows and K6.
Quite probably related, I'm unable to get the software that this driver is based on to work:
https://github.com/RBEGamer/K3_LASER_ENGRAVER_PROTOCOL/issues/6

mgmax, 2022-08-12:
What do you mean with "this"? Please post an error message to avoid misunderstanding.

mgmax, 2022-08-12:
from the issue you linked, it looks like the communication works for multiple packets/commands (ACKs) and then suddenly dies.  I would guess we are doing something subtly wrong, like not enough delay, or not waiting for some software or hardware flow control (XON-XOFF or RTS-CTS). Looking at the exact port configuration and timing of VIsicut and also of the original software with the the serial port monitor could help us find out this.

theohm, 2025-02-15:
Different exception here. The communication seems to fail after a timeout while connecting. Baud rate is set to 115200. Other serial port parameters cannot be set in VisiCut. Looks like a timeout. Could I help finding the cause when installing NetBeans and debugging the code?

    java.io.IOException: Underlying input stream returned zero bytes
        at java.base/sun.nio.cs.StreamDecoder.readBytes(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.implRead(Unknown Source)
        at java.base/sun.nio.cs.StreamDecoder.read(Unknown Source)
        at java.base/java.io.InputStreamReader.read(Unknown Source)
        at java.base/java.io.Reader.read(Unknown Source)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.waitForACK(K3EngraverDriver.java:306)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendLine(K3EngraverDriver.java:322)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.send4ByteCommand(K3EngraverDriver.java:331)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendConnectSequence(K3EngraverDriver.java:425)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendOrWriteJob(K3EngraverDriver.java:588)
        at de.thomas_oster.liblasercut.drivers.K3EngraverDriver.sendJob(K3EngraverDriver.java:568)
        at de.thomas_oster.visicut.VisicutModel.sendJob(VisicutModel.java:740)
        at de.thomas_oster.visicut.gui.MainView.lambda$executeOrSaveJob$7(MainView.java:2143)
        at java.base/java.lang.Thread.run(Unknown Source)