t-oster/LibLaserCut #178: setKeysMissingFromDeserialization() is never called

mgmax, 2021-12-09:
Okey I give up looking. I spend about 90 minutes going through the code and finally I see this section in GenericGcodeDriver.java:1271 after reading all visicut.managers.

```
  /**
   * Adjust defaults after deserializing driver from an old version of XML file
   */
  @Override
  protected void setKeysMissingFromDeserialization()
  {
    System.out.println("Testing whether we ever reach setKeysMissingFromDeserialization:");
    System.out.println("this.spindleMax: " + this.spindleMax);
    System.out.println("this.gCodeDigits: " + this.gCodeDigits);
    System.out.println("this.sCodeDigits: " + this.sCodeDigits);
    // added field spindleMax, needs to be set to 1.0 by default
    // but xstream initializes it to 0.0 when it is missing from XML
    if (this.spindleMax <= 0.0) this.spindleMax = 1.0;
    if (this.gCodeDigits <= 0) this.gCodeDigits = 6;
    if (this.sCodeDigits <= 0) this.sCodeDigits = 0;
  }
```

But according to the debugger this code is never reached and "not part of a class", my printouts don't show up and the values are not being used. Where is this usually handled if not there? I tried debugging this as well as running this as normal and it won't show up with gCodeDigits set to 6. What am I missing?

_Originally posted by @McNugget6750 in https://github.com/t-oster/LibLaserCut/issues/175#issuecomment-987554165_

t-oster, 2021-12-10:
It is documented here https://github.com/t-oster/LibLaserCut/blob/360f95055ec7db490d5598a7eee3dac65951efb8/src/main/java/de/thomas_oster/liblasercut/LaserCutter.java#L431

Maybe the method has to be declared in the implementing class also to work? Not sure, I will test it when I have time.

t-oster, 2021-12-10:
Seems to be introduced by @madleech six years ago. Not sure if it is used anywhere but the generic gcode driver. However one can also use a nullable type and check for null in the getter to achieve the same result