t-oster/LibLaserCut #189: Cut Start with Ruida starts from the origin, instead of first vertex

jnweiger, 2023-03-29:
Seen with Visicut 1.9.0 compiled with liblasercut from https://github.com/t-oster/LibLaserCut/pull/186

The attached SVG should produce a regular 6 pointed star. 
The preview in visicut shows the star correctly, 
but when sent to the laser, the first line starts at the top left origin, instead of at the first vertex.
It is shown incorrect in the machine display, and it also cuts incorrectly.

[image]
[image]

t-oster, 2023-03-30:
Hi,
does this also occur if you compile without #186. Is the problem only in the ruida driver, or can you confirm it in other drivers?

jnweiger, 2023-03-30:
ah, good question!
I should have a chance to verify that with the zing driver soonish...

kkaempf, 2023-03-31:
Hmm, disassembling the generated `.rd` file from `Ruida.java`:

```
...
Blow_on							ca 01 13 
Speed_C9 0.1mm/s					c9 02 00 00 00 00 64 
Laser_1_Min_Pow_C6_01 10%				c6 01 0c 66 
Laser_1_Max_Pow_C6_02 0%				c6 02 00 00 
CA_03 0f						ca 03 0f 
CA 10 00						ca 10 00 
Mov_Abs 49.222mm 46.213mm				88 00 00 03 00 46 00 00 02 69 05 
Cut_Abs 31.351mm 40.147mm				a8 00 00 01 74 77 00 00 02 39 53 
Cut_Abs 22.541mm 56.837mm				a8 00 00 01 30 0d 00 00 03 3c 05 
```

comparing this with https://github.com/t-oster/LibLaserCut/blob/bc3f41385535869d6c47f158ddb7fdb48a2997dd/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java#L585-L595
I'd say that the initial MOVE is coming from LibLaserCut.

t-oster, 2023-04-01:
It looks like the code does make no difference between moveto and lineto commands...lineto commands should be without laser on

jnweiger, 2023-04-01:
Hmm, I doubt that a simple moveto lineto confusion is the issues. If so, we
should see one more lines than expected. But we see exactly 12 line
segments.

kkaempf, 2023-05-10:
Just tested on Nova35 - everything's fine.

A Java version issue (again ?!) 😱 

My environment:
~~~
> java -version
openjdk version "17.0.7" 2023-04-18
OpenJDK Runtime Environment (build 17.0.7+0-suse-1.1-x8664)
OpenJDK 64-Bit Server VM (build 17.0.7+0-suse-1.1-x8664, mixed mode, sharing)
> javac -version
javac 17.0.7
~~~

mgmax, 2023-05-10:
Can you please compare the resulting RD files from the good and bad version?

kkaempf, 2023-05-10:
> It looks like the code does make no difference between moveto and lineto commands...lineto commands should be without laser on

No worries, it does :wink:
https://github.com/t-oster/LibLaserCut/blob/bc3f41385535869d6c47f158ddb7fdb48a2997dd/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java#L595

fblaese, 2023-12-02:
The issue probably is that the last laser position is remembered. If the destination is equal to this position, no move is generated. The remebered position is assumed to be [0,0] on first instantiation. However, this assumption fails in two cases:
- The laser is not positioned at the origion at the start of a program
- The machine code is generated by VisiCut without a restart. In this case, the class instance is reused and the last position of the previous program is used. However, the controller moves the laser back to the origin on program end automatically

I will send a Pull Request to fix this issue soon.

closed 2023-12-09 by mgmax