t-oster/LibLaserCut #194: Ruida driver does not change between Mark and Cut

jnweiger, 2023-09-25:
@kkaempf
Configure green lines for Mark (weak laser power and fast move, to scratch only the surface)
Configure red lines for Cut (strong and slow)

Load an svg that uses both, e.g. the attached head.svg 
[image]

When Mark is first in the list, then everything is marked.
When Cut is first in the list, then everything is cut.

Expected behaviour: when done with one setting, we correctly switch to the other.

jnweiger, 2023-11-09:
The initialization of the first layer "green, mark" is
```
Laser_1_Min_Pow_C6_31 Layer:0 10%                       c6 31 00 0c 66 
Laser_1_Max_Pow_C6_32 Layer:0 35%                       c6 32 00 2c 66
Layer_Speed Layer:0 200.0mm/s                           c9 04 00 00 00 0c 1a 40
Laser_Freq Laser1 00 0.5kHz                             c6 60 00 00 00 00 00 03 74
Layer_Color Layer:0 R 39%, G 0%, B 0%                   ca 06 00 00 00 00 00 64 
Layer_CA_41 Layer:0 00                                  ca 41 00 00 
Flags_CA_01 00                                          ca 01 00 
CA 02 Prio 0                                            ca 02 00
Blow_on                                                 ca 01 13 
Speed_C9 0.2mm/s                                        c9 02 00 00 00 01 48 
Laser_1_Min_Pow_C6_01 10%                               c6 01 0c 66 
Laser_1_Max_Pow_C6_02 35%                               c6 02 2c 66
CA_03 0f                                                ca 03 0f
CA 10 00                     
```

The initialization of the second layer "red, cut" is
```
Laser_1_Min_Pow_C6_01 50%                               c6 01 3f 7f
Laser_1_Max_Pow_C6_02 80%                               c6 02 66 33
Speed_C9 14.0mm/s                                       c9 02 00 00 00 6d 30
```

jnweiger, 2023-11-09:
With "my" old visicut 1.8, th initialization looked like this:
First layer: 
```
Layer_Speed Layer:0 200.0mm/s                           c9 04 00 00 00 0c 1a 40
Laser_1_Min_Pow_C6_31 Layer:0 9%                        c6 31 00 0b 42
Laser_1_Max_Pow_C6_32 Layer:0 30%                       c6 32 00 26 33
Layer_Color Layer:0 R 0%, G 0%, B 0%                    ca 06 00 00 00 00 00 00
Layer_CA_41 Layer:0 00                                  ca 41 00 00
Layer_Top_Left_E7_52 Layer:0 0.0mm 0.0mm                e7 52 00 00 00 00 00 00 00 00 00 00 00
Layer_Bottom_Right_E7_53 Layer:0 20.066mm 15.087mm      e7 53 00 00 00 01 1c 62 00 00 00 75 6f
Layer_Top_Left_E7_61 Layer:0 0.0mm 0.0mm                e7 61 00 00 00 00 00 00 00 00 00 00 00
Layer_Bottom_Right_E7_62 Layer:0 20.066mm 15.087mm      e7 62 00 00 00 01 1c 62 00 00 00 75 6f
Layer_Speed Layer:1 30.0mm/s                            c9 04 01 00 00 01 6a 30
Laser_1_Min_Pow_C6_31 Layer:1 50%                       c6 31 01 3f 7f
Laser_1_Max_Pow_C6_32 Layer:1 70%                       c6 32 01 59 4c
Layer_Color Layer:1 R 100%, G 0%, B 0%                  ca 06 01 00 00 00 01 7f
Layer_CA_41 Layer:1 00                                  ca 41 01 00
Layer_Top_Left_E7_52 Layer:1 0.0mm 0.0mm                e7 52 01 00 00 00 00 00 00 00 00 00 00
Layer_Bottom_Right_E7_53 Layer:1 30.073mm 20.066mm      e7 53 01 00 00 01 6a 79 00 00 01 1c 62
Layer_Top_Left_E7_61 Layer:1 0.0mm 0.0mm                e7 61 01 00 00 00 00 00 00 00 00 00 00
Layer_Bottom_Right_E7_62 Layer:1 30.073mm 20.066mm      e7 62 01 00 00 01 6a 79 00 00 01 1c 62
Flags_CA_01 00                                          ca 01 00
CA 02 Prio 0                                            ca 02 00
Blow_on                                                 ca 01 13
Speed_C9 200.0mm/s                                      c9 02 00 00 0c 1a 40
Laser_1_Min_Pow_C6_01 9%                                c6 01 0b 42
Laser_1_Max_Pow_C6_02 30%                               c6 02 26 33
CA_03 0f                                                ca 03 0f
CA 10 00                                                ca 10 00
```

Second layer:
```
Flags_CA_01 00                                          ca 01 00
CA 02 Prio 1                                            ca 02 01
Blow_on                                                 ca 01 13
Speed_C9 30.0mm/s                                       c9 02 00 00 01 6a 30
Laser_1_Min_Pow_C6_01 50%                               c6 01 3f 7f
Laser_1_Max_Pow_C6_02 70%                               c6 02 59 4c
CA_03 0f                                                ca 03 0f
CA 10 00     
```

jnweiger, 2023-11-10:
```
diff --git a/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java b/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java
index 57292fa..d9dc425 100644
--- a/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java
+++ b/src/main/java/de/thomas_oster/liblasercut/drivers/Ruida.java
@@ -588,6 +588,9 @@ public class Ruida extends LaserCutter
     boolean first_prop = true;
     boolean first_vector = true;
 
+    // FIXME: part_number is never incremented.
+    // FIXME: first_vector and first_prop should be reset after layer change. No?
+
     for (JobPart p : job.getParts())
     {
       float focus;

fblaese, 2023-11-10:
See https://github.com/fblaese/LibLaserCut/commit/d36821b5df5e8bc871097d2a627429983e50f029 for a possible fix. The part_number increment and first_* reset for each part probably was lost at some point in time.

Compared to before, the Layer_* commands are now in front of each layer's code instead of the beginning. This does not seem to be an issue, though.

closed 2023-12-09 by mgmax