t-oster/LibLaserCut #207: grblHAL http support

pimsierhuis, 2024-02-01:
We'd like to use [grblHAL](https://github.com/grblHAL/) for our laser cutters so we started rewiring one to use the [picocnc](https://github.com/phil-barrett/PicoCNC) board with a pi pico. It would be nice if a file is first uploaded to an SD-card and then streamed from SD instead of streaming directly over the network (because of reliability). So we like to use the networking-plugin of grblhal and the http-server for uploading files.

The issue is, that the GenericGcodeDriver uses a different approach for http-upload (POST with X-filename header) than grblhal uses. See [here](https://esp3d.io/esp3d-webui/v3.x/documentation/api/fileupload/index.html), implemented [here](https://github.com/grblHAL/Plugin_networking/blob/master/http_upload.c#L97)

So I want to propose a setting for the GenericGcodeDriver modifying the behaviour of the http_upload-method. I'm willing to create a PR for that. Would this be acceptable? Or do you have other suggestions?

pimsierhuis, 2024-02-01:
Oh, and then the question from #196 pops up; maybe a good moment to introduce a new httpclient? If yes, which? Would it be a problem to require java 11, so being able to use a [native httpclient](https://openjdk.org/groups/net/httpclient/intro.html)? Or should we use for example [Apache httpclient](https://hc.apache.org/)

t-oster, 2024-02-01:
Hi, in my opinions pull reqests are always welcome, as far as they don't break existing settings. So if you make this configurable in GenericGcodeDriver make sure the default is the old behavior and old settings are imported correctly with that default.

We are already on java 11, so the native client would be preferrable if we get it to work without too much hassle.

pimsierhuis, 2024-02-01:
Thanks for the fast reaction, will start working on it and try to implement it with the native http client

PR #220 (2024-04-19): GenericGcodeDriver: Add support for grblHAL HTTP-upload and static jo…
…b name. Fixes #207
commit 365301a0 (2024-04-19): GenericGcodeDriver: Add support for grblHAL HTTP-upload and static job name. Fixes #207
closed 2024-04-19 by mgmax