t-oster/LibLaserCut #211: Support additional dithering algorithms

TheAssassin, 2024-03-03:
I'm not an expert on image processing, but the current list seems a bit limited. When looking, e.g., at the vendor-provided Epilog Zing printer driver, it supports Jarvis and Stucki (the latter seems like a progression of the Floyd-Steinberg algorithm that is already available).

I would welcome some additional algorithms to choose from, though.

This is not a big priority and I'm not sure how complex adding those would be.

mgmax, 2024-03-03:
I'd like to add one bit of experience here: Many dither algorithms are good in theory on computer screens but ugly in practice on laser cutters because the laser slightly "widens" the engraved pixel. So one black pixel becomes like 2x2 pixels. This means that not all algorithms are suitable. Also, gamma correction is also very important, which motivated the "Brightened halftone" dither option.

Attached are some microscope pictures (height measurement and optical view) of engraving a filled rectangle with 500dpi with Epilog Zing with a very aged laser source. Two rows were partially missing due that laser tube problem. Unfortunately I don't have pictures of other engravings, but I could try to get some new ones if we have a specific need (e.g., a new proposed algorithm and/or a good pattern for testing).

Note that the images show about 1mm in the Y direction, which are ca. 20 engrave rows at 500 DPI. (The boundary of the engraving looks only like about 10 rows due to thermal effects.)  My guess is that one "engrave pixel" is actually 1.5 - 2 pixels wide.

Optical view, probably with some kind of polarization filter:
[image]

Height profile:
[image]
[image]
[image]

(note to self: the soure email is "Microscope images" from 2027-10-23)

TheAssassin, 2024-03-03:
Just to test my understanding: Given the laser widens the pixel(s) means that some more pixel binning like Stucki for instance apparently does could improve the result then?

TheAssassin, 2024-03-03:
That sounds like a good idea. Decisions based on research are always well-founded.

By the way, I have a stereo microscope at hand, so I could inspect (and photograph) results on the Epilog Zing and another device and provide feedback.